udacity/robotics-nanodegree-issues #2: Radio button colors

karrapk, 2017-05-24:
Hello, Radio button colors and input text box colors are very close to the background color of Udacity class pages. I am finding it quite a bit difficult to distinguish various fields and their corresponding options in Udacity quizzes. Could you please differentiate the above fields a bit more so it becomes easier? I have tried two monitors and I faced the issue using both. Thanks.

ryan-keenan, 2017-05-25:
Thanks for this feedback!  Could you please post a screenshot of an example or two where you're having issues?  I'm not sure exactly what the change requested is.

karrapk, 2017-05-25:
Hello,

I am attaching the mobile version of the screen shot here as I am away from
the PC. If you go to the PC version of the attached page, the button where
the student is supposed to make the selection is very feebly visible due to
their color being very similar to the background color the page.

This makes it very difficult to identify the right buttons or boxes where
sometimes data needs to be entered. I can reply to this email tomorrow with
PC version screen shot if you like.

Thanks,
Prashanth.

ryan-keenan, 2017-05-25:
Thanks Prasanth, I don't see any image though...

karrapk, 2017-05-25:
I attached a png in the previous email. Not sure if the email client
removed it. Please see the attached jpg. I rounded the radio buttons in
red. Thanks.

sahiljuneja, 2017-05-25:
@karrapk I don't think images attached via email are attached here as well. You might have to attach it separately on github perhaps.

karrapk, 2017-05-25:
[image]

Screenshot attached.

ryan-keenan, 2017-05-25:
Thank you for the image.  I have filed a ticket with our platform team but it will probably not be addressed soon as they have many other issues in their queue.  Sorry for the inconvenience!

closed 2017-05-25 by ryan-keenan