udacity/robotics-nanodegree-issues #4: Jupyter still needs plt.show()

Dulluhan, 2017-05-25:
There is a code block in Lesson 1 Project.6.Persepective Transform that has comment saying plt.show() should not be needed in Jupyter, however, in my case it seems that it doesn't show unless I have that. It is the same case if I was to copy the previous section's code to try out on my own computer. It would be helpful to let people know that to view the plt object locally, .show() is needed.

Dulluhan, 2017-05-25:
https://puu.sh/w0rjP/a5878a6496.png

ryan-keenan, 2017-05-25:
Thanks for reporting this!  As far as I know plt.show() has no adverse effects even if it's not needed (not sure why some notebooks seem to need it others don't) so I've gone through and deleted the comments saying `# not needed in Jupyter notebook`

closed 2017-05-25 by ryan-keenan