udacity/robotics-nanodegree-issues #6: cv2 library install

Dulluhan, 2017-05-25:
Using RoboND, cv2 import gets error: ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
Unsure if it is a problem because I have kinetic setup locally, but RoboND should account for this?

Dulluhan, 2017-05-25:
Seems like its an issue with the path being added by ROS. Solution: https://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv
Maybe good to account for this in later stages since ROS will be an integral part of this course

stevekim1357, 2017-05-25:
I am having problem getting matplotimag lib to load to conda python

Dulluhan, 2017-05-25:
Hey, matplotlib worked for me, maybe you want to run $conda install for that package and try again. Use conda instead of pip so conda adds it to the workspace

stevekim1357, 2017-05-26:
thanks I will try it today Sincerely, Steve W Kim stevekim_1357@yahoo.com

stevekim1357, 2017-05-26:
well, I don't know where to turn to, am I suppose to run it on my computer? Sincerely, Steve W Kim stevekim_1357@yahoo.com

Chambana, 2017-05-27:
I have ROS Kinetic installed in my native Ubuntu environment.  Per your suggestion, I modified my bashrc and commented out the ROS Path stuff.  The Jupyter Notebook for the Udacity exercise now runs.  

For future changes to the exercise/environment down the line, it would be great if the exercises still worked for students that have ROS installed on their machines w/o having to disable ROS path in the bashrc file.

Dulluhan, 2017-05-28:
@stevekim1357 Try running that on your computer, try `$ conda install matplotlib`

ryan-keenan, 2017-06-17:
Thanks for reporting these issues.  I'm going to close this now as I believe the initial problem is solved in the [doc](https://github.com/udacity/RoboND-Python-StarterKit/blob/master/doc/linux_ros_anaconda_warning.md) that's been added to the environment setup repo.  Please reopen if problems persist.

closed 2017-06-17 by ryan-keenan