udacity/robotics-nanodegree-issues #8: roversim in non-english locale

TillScout, 2017-05-25:
The roversim produces a csv-file that uses "," as delimiter. However this is also the decimal mark in e.g. German. So when reading the csv-file, the values of all float variables are spread across several columns

ryan-keenan, 2017-05-25:
Great point!  Are you using `pandas` to read the file like in the notebook?  If so I think it should work ok

TillScout, 2017-05-25:
I only noticed it when checking out the rover sim. Then I changed my locale, so I can't say how `pandas` behaves.

ryan-keenan, 2017-05-25:
Ok, I think the way the notebook is setup it shouldn't be a problem so I'll close this for now, but feel free to reopen if there is a persistent problem here.

closed 2017-05-25 by ryan-keenan

TillScout, 2017-05-26:
I now tried it with `pandas` in the notebook. This gave an error, because `pandas`expects `,`as a standard delimiter. On my system the roversim used `;`as delimiter (it was the newer version, maybe you changed something there). So I have the feeling that the sim and the notebook are out of sync in this regard.