udacity/robotics-nanodegree-issues #9: Simulator crashes when I press enter and am near an objective.

mi7flat5, 2017-05-25:


ryan-keenan, 2017-05-25:
Thanks for reporting this... which OS are you on?  and when did you download the Sim player?

mi7flat5, 2017-05-25:
I'm on Windows 10 64-bit, I downloaded the sim about an hour after the course opened.

ryan-keenan, 2017-05-25:
Ok, I'll look into this.   I updated the sim with mostly aesthetic changes a few hours after the course opened so you can try the new version and see if there's any change, but I'll follow up and see if I can reproduce the error meanwhile.

ryan-keenan, 2017-05-25:
In beta testing this issue arose and seemed to resolve itself after a restart of the machine and simulator.  Could you try a reboot and see if it helps?

mi7flat5, 2017-05-25:
Still crashes. I have the project open in Unity and get a null reference error on line 598 of RoverController.cs when I pick up the item. Unity still goes through the animation in the editor. Looks like curObjective is set to null in line 590, and removal is attempted in 598, perhaps this would explain the crash?

ryan-keenan, 2017-05-25:
Yeah, just playing with this and it looks like a bug.  Left mouse button should work fine for pickup now and I'll upload a fixed version of the sim as soon as I can!

mi7flat5, 2017-05-25:
I just fixed it by moving line 590 to just before the yield statement. I don't know the logic in the if(IsRecording) statement, but seems to run fine now.

ryan-keenan, 2017-05-25:
This has been fixed now!  Repo updated and new builds on S3

closed 2017-05-25 by ryan-keenan