udacity/robotics-nanodegree-issues #15: Issues in Part 10) - Sourcing the Environment of the Introduction to ROS lesson

adammalpass, 2017-05-30:
just want to report two issues I found in Part 10) - Sourcing the Environment of the Introduction to ROS lesson:

1) In the text it mentions  "/.bash.rc file". This is incorrect, it should be  "/.bashrc", i.e. no 'dot' between 'bash' and 'rc'.

2) The exercise talks about adding the sourcing of the ROS bash script to the bashrc so we can load everytime and walks through an exercise to compare the difference. However within the VM Udacity provided the sourcing of the ROS scripts is already included in the bashrc file. So it is not necessary to add it again, and furthermore the diff between oldenv and newenv returns nothing (since nothing has changed since it was already sourced). I think this could be quite confusing for new students.

Please can you update the instructions? Thanks!

kylesf, 2017-06-09:
Hey there Adam, 

Thanks for pointing out the `/.bash.rc file`, I went ahead and fixed that. As for #2, I agree that it can be made clear that the action does not need to be made on the VM. I will look into that. However, some may not be using the supplied VM and therefore this constitutes as a more complete walkthrough on the appropriate steps. As for the `diff between oldenv and newenv returns nothing (since nothing has changed since it was already sourced)` matter, the classroom actually says this `One quick way to determine what changes were made to the existing environment is to run the following commands in a new terminal (where `/opt/ros/kinetic/setup.bash` has not already been sourced).`  Do these help alleviate any confusion?

kylesf, 2017-06-10:
Made a note in the classroom informing students that they do not need to add the sourcing of ROS to their bashrc file since it is already taken care of!

closed 2017-06-10 by kylesf