udacity/robotics-nanodegree-issues #17: Bug in the UI for the quizzes

sahiljuneja, 2017-05-30:
Whenever you submit a quiz, you get a pop-up window stating whether we cleared it or not.

If you press the `Try Again` or `Continue` button in that pop-up, there's no issue.

But if you try to close the pop-up by clicking the `x` (cross) in the top-right corner of the pop-up window, it scrolls the window all the way to the top.

ryan-keenan, 2017-05-31:
filed a ticket for this, thanks for reporting!

kylesf, 2017-06-29:
@ryan-keenan Any updates on the JIRA ticket?

kylesf, 2017-08-02:
@sahiljuneja Did this ever get fixed to your knowledge?

sahiljuneja, 2017-08-02:
Nope. Just checked. The JIRA ticket is unresolved (in review) and they will get to it I'm sure. So I will close this since it's in the system and is not of high priority.

closed 2017-08-02 by sahiljuneja