udacity/robotics-nanodegree-issues #20: Lesson 3 - Part 3 - Simple Mover The Code

adammalpass, 2017-05-31:
In the full code listing there is no call to 'rate.sleep()' inside the first while loop 'while not start_time', however there is one inside the code breakdown. Based on the description underneath the code explanation I think this needs to be added to the main code listing - please confirm?

kylesf, 2017-06-09:
Hey there Adam, I am not sure anything needs to change but lets chat about it and see if we can make sense of it! 

So the main repeating part of the overall code is this while loop:
```
while not rospy.is_shutdown():
        elapsed = rospy.Time.now().to_sec() - start_time
        pub_j1.publish(math.sin(2*math.pi*0.1*elapsed)*(math.pi/2))
        pub_j2.publish(math.sin(2*math.pi*0.1*elapsed)*(math.pi/2))
        rate.sleep()
```

The ` rate.sleep()` is to maintain the loop cycle rate. As for the while loop above it `while not start_time` is only called upon during the initialization of this file. It is needed because of the try and except functions in the init part of the code that would want to take advantage of the previously running time.

adammalpass, 2017-06-10:
Hi Kyle, it looks like the issue has already been fixed. Previously on the lesson page there was no rate.sleep() in the first while loop in the entire code listing, but underneath where it repeats sections of the code again for the breakdown there _was_ a rate.sleep() line. So my point was that the full code and the code breakdown were inconsistent. But now it's fixed so I think this can be closed, thanks!

closed 2017-06-10 by kylesf