udacity/robotics-nanodegree-issues #25: Lesson 3 - Part 8 - Code error

frozenamazon, 2017-06-03:


frozenamazon, 2017-06-03:
In the extra functions.py provided, img.shape[0] and img.shape[1] has been swapped. Took me hours to troubleshoot, but at the end the LiveHelp saved me :) Thanks Admin!

Provided: 
    dst = np.float32([[img_size[0]/2 - dst_size, img_size[1] - bottom_offset],
                      [img_size[0]/2 + dst_size, img_size[1] - bottom_offset],
                      [img_size[0]/2 + dst_size, img_size[1] - 2*dst_size - bottom_offset], 
                      [img_size[0]/2 - dst_size, img_size[1] - 2*dst_size - bottom_offset],
                      ])

Should be: 
  dst = np.float32([[image.shape[1]/2 - dst_size, image.shape[0] - bottom_offset],
                  [image.shape[1]/2 + dst_size, image.shape[0] - bottom_offset],
                  [image.shape[1]/2 + dst_size, image.shape[0] - 2*dst_size - bottom_offset], 
                  [image.shape[1]/2 - dst_size, image.shape[0] - 2*dst_size - bottom_offset],
                  ])

ryan-keenan, 2017-06-10:
Thanks for reporting this!  Indeed there was an inconsistency between this and the perspective transform quiz.  Fixed now so they are the same code.

closed 2017-06-10 by ryan-keenan