udacity/robotics-nanodegree-issues #26: Simulator creating large file under ~/.config

Dulluhan, 2017-06-04:


kylesf, 2017-06-09:
Could you please provide more information in regards to your OS and the version of the simulator you are using? Also the size of `~/.config`. Thanks!

Dulluhan, 2017-06-11:
Hi, I'm on Ubuntu 16.04, Roversim.x86_64, I think the problem has since been updated and partially fixed. It does still add 8mb every time I restart the node.
At the time I reported this I had a 16gB folder under `~/.config/unity3d/`, I  added `rm -rf ~/.config/unity3d/` to my `.bashrc` so I have not had that problem since then. I noticed this problem because my computer was constantly telling me I was running out of disk space in `/` and I couldnt figure out why

Dulluhan, 2017-06-11:
I did some more testing, seems that the folder grows when the python node is inactive, maybe the simulator is constantly creating logs when its not connected to python socket? Its grown from 19mb to almost 1gb in 5 minutes when my python node isn't open

Dulluhan, 2017-06-11:
The problem does go away after the python node is restated

Dulluhan, 2017-06-11:
Its under `~/.config/unity3d/Rover Simulator/Rover Simulator/Player.log`

Dulluhan, 2017-06-11:
Its a connection refusal in with sockets, building error logs. 
```
11/06/2017 3:53:46 PM|Fatal|WebSocket.acceptException|System.Net.Sockets.SocketException: Connection refused
                              at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP, Boolean requireSocketPolicy) [0x00000] in <filename unknown>:0 
                              at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP) [0x00000] in <filename unknown>:0 
                              at System.Net.Sockets.TcpClient.Connect (System.Net.IPEndPoint remote_end_point) [0x00000] in <filename unknown>:0 
                              at System.Net.Sockets.TcpClient.Connect (System.Net.IPAddress[] ipAddresses, Int32 port) [0x00000] in <filename unknown>:0 
```

kylesf, 2017-06-11:
Do the logs auto delete at all? If not perhaps we can build this in as a feature

kylesf, 2017-06-16:
Any updates to this?

Dulluhan, 2017-06-17:
For me, the have just been building up

kylesf, 2017-06-18:
Can you see if the problem occurs with the newest simulator version, this could be related to a memory leak issue we had previously.

kylesf, 2017-06-29:
Status update?

closed 2017-07-19 by kylesf