udacity/robotics-nanodegree-issues #28: Is picking_up flag does not reset in simulator

Dulluhan, 2017-06-11:
After pick_up flag is sent through socket, picking_up flag is asserted on Unity end. The result of this is that the rover no longer takes any commands manual or autonom4 a sample nearby. Could be just a problem with my build. Ubuntu 16.04, Roversim.x86_64

Dulluhan, 2017-06-11:
After pick_up flag is sent through socket, picking_up flag is asserted on Unity end. The result of this is that the rover no longer takes any commands manual or autonoms, with or without a sample nearby. Could be just a problem with my build. Ubuntu 16.04, Roversim.x86_64

kylesf, 2017-06-11:
Is this with the most up to date simulator?

kylesf, 2017-06-16:
Any updates to this?

ryan-keenan, 2017-06-17:
I'll close this issue for now as this should be fixed in the most recent updates to the project repo.  Please reopen if problems persist!

closed 2017-06-17 by ryan-keenan