udacity/robotics-nanodegree-issues #33: Two DoF Arm Code - Bug

adammalpass, 2017-06-14:
There seems to be a bug in the Two DoF Arm code quiz. The comments in the code states:

"NOTE: joint2_angle is the angle counterclockwise from the link1 axis"

However the code solution.py (line 6&7) specifies relative to the x-axis (joint1_angle + joint2 angle). When you press 'Submit Answer' it also only marks you correct if your code has (joint1_angle + joint2 angle). However looking at the arm plot it is clear that really the joint2_angle is relative to link1_axis (as per code comment) and if you set (joint1_angle + joint2 angle) then the plot is wrong (but marked correct) and if you set to just joint2_angle then the plot looks correct (but is marked wrong).

I think both the solution grader and the solution.py need updating.

adammalpass, 2017-06-14:
Sorry I jumped the gun a bit on this one... my mistake... code is correct, I was wrong. Please close.

sahiljuneja, 2017-06-14:
Just saw the Slack discussion. Glad it was cleared out. :)

closed 2017-06-14 by sahiljuneja