udacity/robotics-nanodegree-issues #35: RoboND-Kinematics-Project README mentions wrong repo to clone

adammalpass, 2017-06-16:
In the README (https://github.com/udacity/RoboND-Kinematics-Project/tree/435619d7e7ff67ea4ad469ad3ced64b5acc96070) of the RoboND-Kinematics-Project it says we should clone the following repo:

```$ git clone https://github.com/udacity/test_repo_robond_robotic_arm_pick_and_place.git```

However in the lesson notes in Part2: Setting Up Your Environment it says we should clone this repo:

```$ git clone https://github.com/udacity/RoboND-Kinematics-Project.git```

I think the lesson notes are correct and the README instructions are out of date.  Please can you confirm and if so update the README? Thanks!

sahiljuneja, 2017-06-16:
Thanks a lot for pointing this out! Should be updated soon once the Bay Area (California) wakes up ;)

sahiljuneja, 2017-06-16:
Should be fixed now.

closed 2017-06-16 by sahiljuneja