udacity/robotics-nanodegree-issues #42: Diagram needs revision

irishMee, 2017-06-17:


irishMee, 2017-06-17:
in the Homogeneous Transforms and their Inverse lesson and in the first diagram on the page, I believe that the vector labeling for A rAo/Bo and the vector  r P/Ao need to be switched. Image that I am talking about is shown below with my recommended switch.
[image]

ryan-keenan, 2017-06-17:
Thanks for reporting!  This looks like [issue #38]( https://waffle.io/udacity/robotics-nanodegree-issues/cards/594446fbe8304a00263d6354) so will assign to @jchernus as well and move to "in progress".

closed 2017-06-29 by jchernus