udacity/robotics-nanodegree-issues #43: rqt_image_view throws segmentation fault

Chambana, 2017-06-17:
Every time I run rqt_image_view in the Arm Mover module it throws a segmentation fault.  I'm running Ubuntu 16.04 and ROS Kinetic.  Just did a fresh reinstall of ROS Kinetic Desktop Full and it's still happening.  Any fix would be appreciated!

kylesf, 2017-06-18:
Is this happening via the supplied VM?

Chambana, 2017-06-18:
No.  I'm just using 16.04 LTS and Kinetic.

sahiljuneja, 2017-06-18:
This is a bit difficult for us to help with since we can't easily reproduce the error in the VM (which is what's officially supported for this course currently). 

But such a problem is usually associated with not having an up-to-date QT framework.

So you can check your version by running `qmake --version`

For Kinetic the minimum version stated here http://wiki.ros.org/rqt is 5.3, although for the VM we currently have 4.8.7

So you can try updating that if there is a mismatch. Here are some more references on this matter -

- https://github.com/ros/rosdistro/issues/11488
- http://answers.ros.org/question/239043/rqt_image_view-segmentation-fault/

This could also be related to the OpenCV version -
- http://answers.ros.org/question/212846/cv_bridgeopencv-segmentation-fault/?answer=241928#post-id-241928

But beyond this, it might be a bit more difficult for us to help out. There are many students who have their local builds up and running. You can discuss this issue with them in the course Slack in the #ros channel. 

Let us know if any of the above helps and we can see how we can help further!

closed 2017-07-03 by sahiljuneja