udacity/robotics-nanodegree-issues #46: Crashy gazebo

mi7flat5, 2017-06-19:
4 out of 5 attempts to run safe_spawner.py results in a gazebo crash. This occurs in either mode. I've attempted increasing the VM memory to 8GB. Still the same. I get one of two error, if any error at all.

 [gazebo_gui-6] process has died [pid 94347, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/robond/.ros/log/b3006d02-5526-11e7-be26-000c296a8714/gazebo_gui-6.log].
log file: /home/robond/.ros/log/b3006d02-5526-11e7-be26-000c296a8714/gazebo_gui-6*.log

[gazebo-5] process has died [pid 88740, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/robond/catkin_ws/src/RoboND-Kinematics-Project/kuka_arm/worlds/cafe.world __name:=gazebo __log:=/home/robond/.ros/log/df098fa8-5523-11e7-be26-000c296a8714/gazebo-5.log].
log file: /home/robond/.ros/log/df098fa8-5523-11e7-be26-000c296a8714/gazebo-5*.log

sahiljuneja, 2017-06-19:
That's interesting. It shouldn't crash that often. 

Can you try to open that log and sharing what error the log shows? The path is presented in the error above.

Also, just as a check. In your VMWare settings (at least the one for Windows), under "Display" there might be an option of "Accelerate 3D graphics" Can you confirm if that's checked or not? It should ideally be UNchecked and that might help with gazebo, although not guaranteed.

mi7flat5, 2017-06-19:
The "Accelerate 3D graphics" was checked. I unchecked it, and now Gazebo starts in a much smaller window, but still crashes. I check the log file and it is empty.

bkinman, 2017-06-21:
Have you tried `$ ps aux | grep gz` after the crash, to verify that there are no instances of gazebo client or server running?

Can you try waiting 30 seconds after the crash and then trying again? I know it sounds odd, but I have experienced situations  with gazebo where waiting a full 30 seconds before relaunching made this type of crash go away.

I actually have an open ticket with OSRF to investigate the random crashes on starting, but I've not received any response from them yet.

mi7flat5, 2017-06-21:
Yes there does seem to be a time factor at play here, if I wait the odds of success increase I haven't confirmed that another instance is still running, I assume this is part of the issue. I changed the settings in the VM to 6 cores and gave it 13GB of RAM and gave the video 2gb and it seems to be working better as well. Though sometimes still a crash, not nearly as often now.

bkinman, 2017-06-25:
I actually run into this problem on my system as well, and I'm not using a VM.

kylesf, 2017-06-29:
Do we have updates to this matter?

hvpandya, 2017-07-26:
This is a gazebo issue (have been experiencing this for years now) and not something we can resolve. 

`gzserver` is the culprit most of the times as it keeps hanging around after a crash, waiting for certain processes to die elegantly.

As I mentioned before, this is not something we can resolve, hence closing the ticket.

closed 2017-07-26 by hvpandya