udacity/robotics-nanodegree-issues #47: Pick and Place - Arm Doesn't move to dropbox

adammalpass, 2017-06-19:
I am doing pick&place project and my arm successfully plans a route to the object and picks it. It then successfully plans the route to the dropbox. However the arm does not _move_ to the dropbox. I can see the following errors in the console:

[ERROR] Failed to call service calculate_ik
[ERROR] found empty JointState message

I'm not certain, but it looks to me like these are not issues within my code, but within the code calling the IK_server. Someone others also mentioned this on Slack.

Please can you check?

sahiljuneja, 2017-06-19:
Hmm... I will try to replicate the problem on my end. This might take a while though but I will try to look into it asap.

Is the code in your github repo the latest? I can try that along with our solution to see what might be missing too.

sahiljuneja, 2017-06-19:
When I said, "the code in your GitHub repo" I meant "your solution, the ik_server.py file"

adammalpass, 2017-06-19:
Yes my code in GitHub is the latest.

Problem may have just been resolved via Slack though - safe_spawner.sh doesn't start our code directly, we need to run it separately via rosrun.

adammalpass, 2017-06-20:
Yes it's ok now, thanks!

closed 2017-06-20 by sahiljuneja