udacity/robotics-nanodegree-issues #49: live help button missing in chrome, but present in firefox

flores-jacob, 2017-06-23:
There is no livehelp button on the lower left hand corner of the screen when using the chrome browser.  Firefox however has the button.

Chrome version:Version 59.0.3071.109 (Official Build) (64-bit)
OS: Ubuntu 16.04

kylesf, 2017-07-05:
Tried to replicate your error and was unable to do so. Does this still happen?

flores-jacob, 2017-07-06:
Hi. Yes, it still happens. However, I believe I just found out why.
Apparently, the livehelp button only shows up if the lesson page is opened
with the https:// prefix. Without the https prefix, the livehelp button
does not show up. This is true for firefox and chrome. I had the a firefox
plugin that always forced https connections, which would explain why the
button shows up on firefox, and not on chrome. Maybe Udacity can force
https connections when a student is logged into his or her account. This
would ensure that the button is always visible.

kylesf, 2017-07-06:
Okay will share!

kylesf, 2017-07-25:
We have moved this into internal tickets and assure you its being addressed! Thanks for your input!

closed 2017-07-25 by kylesf