udacity/robotics-nanodegree-issues #54: Robot model URDF not loading correctly into RViz

christiansafka, 2017-06-26:
This issue has been reported by two people both running Ubuntu 16.04.  Other URDF gazebo models are working fine.  When loading this one, only the gripper is visible.  The collisions part seems to load fine, and shows the whole robot in gray.  There is something going on with the visual part of the robot.

sahiljuneja, 2017-06-26:
Is there any error message or warnings or anything that pop-up when you try to launch it? Share a screenshot of the terminal if possible.

Also, are you sure you added the correct path in your `bashrc` for the `gazebo_model_path` and sourced it? Usually that is the problem.

christiansafka, 2017-06-26:
[image]

I had actually forgotten to edit the gazebo_model_path!  Unfortunately, the problem still exists after sourcing the correct one, and it seems that the gazebo model path is only used in gazebo (for the bin, shelf, etc. models).

sahiljuneja, 2017-06-26:
Did you directly copy-paste the gazebo_model_path from the lessons?

Because the workspace in the lessons is `catkin_ws` and yours is `ros_ws`. That difference might lead to such issues.

christiansafka, 2017-06-26:
No, after you mentioned the gazebo model path, I made sure that it had the correct path.  Also, that path points to models for Gazebo simulation, not sure how it would affect this demo.  But you definitely saved me some headaches later :)

sahiljuneja, 2017-06-29:
Hey, is this still troubling you?

I have had a difficult time to reproduce this error on my end.

You don't seem to be working with the classroom VM so difficult to say what might be causing the problem based on anything else you might have installed/downloaded previously.

Couple of things to try would be to remove your current workspace and then follow steps in the lessons to create one from scratch. Or try to create a separate workspace based on the lessons. Although the latter might cause some issues, so I don't recommend that yet.

christiansafka, 2017-07-19:
I was away for a couple weeks, and yes this is still an issue.  Another case of this has just been reported on Slack, and I will ask the person to post specs here.  @kylesf Can you re-open the issue?

kylesf, 2017-07-19:
I will reopen for a few days. We do not offer support for anything other than the provided VM. If you are able to propose a solution to help other students we can share that.

ruffson, 2017-07-25:
Same issue here. Tried everything. At least you can enable the collision model in rviz so that you see some kind of visual...

kylesf, 2017-08-09:
Are there updates to this matter?

closed 2017-08-28 by kylesf

nardo991, 2018-03-02:
Hi, i have the same issue.

I checked if the URDF can be converted into a SDF file. font:http://gazebosim.org/tutorials/?tut=ros_urdf
received these errors. if someone is interested in solving  the issue, maybe this is the way.

> gz sdf -p kr210.urdf.xacro 
> Error:   Inertial: mass [${mass0}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [base_link]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass1}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_1]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass2}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_2]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass3}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_3]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass4}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_4]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass5}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_5]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Inertial: mass [${mass6}] is not a float
>          at line 306 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   Could not parse inertial element for Link [link_6]
>          at line 434 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/link.cpp
> Error:   lower value (${-185*deg}) is not a float: bad lexical cast: source type value could not be interpreted as target
>          at line 119 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/joint.cpp
> Error:   Could not parse limit element for joint [joint_1]
>          at line 494 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/joint.cpp
> Error:   joint xml is not initialized correctly
>          at line 207 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/model.cpp
>