udacity/robotics-nanodegree-issues #55: catkin_make fails after upgrading gazebo

fredrick-dominy, 2017-06-29:


fredrick-dominy, 2017-06-29:
Initial walkthrough installed correctly from the udacity course and I was able to play around with the various applications.  

However when I forked the project, instructions were to upgrade gazebo to 7.7+  (If I remember correctly I was on 7.0 or something). 

Now catkin_make fails. 

"/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory"

Same error after multiple deletions and rebuild cycles.

-------- Error Follows------
[ 91%] Built target trajectory_sampler
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/fredrick/catkin_ws/src/RoboND-Kinematics-Project/gazebo_grasp_plugin/src/GazeboGraspFix.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/fredrick/catkin_ws/src/RoboND-Kinematics-Project/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
compilation terminated.
RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/build.make:62: recipe for target 'RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o' failed
make[2]: *** [RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/build.make:86: recipe for target 'RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o' failed
make[2]: *** [RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o] Error 1
CMakeFiles/Makefile2:463: recipe for target 'RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/all' failed
make[1]: *** [RoboND-Kinematics-Project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

fredrick-dominy, 2017-06-29:
After uninstall of ROS and reinstallation from square-1, catkin_make was able to complete.  

Issue resolved.

PEBCAK

sahiljuneja, 2017-06-29:
Glad it worked! Was there anything different you had to do when reinstalling?

closed 2017-06-29 by sahiljuneja

fredrick-dominy, 2017-06-29:
I ensured that ROS and all configurations were deleted, including the
hidden folders found on ~

Also, I did not upgrade gazebo, as directed from the GitHub repo... I think
the version it's currently 7.8