udacity/robotics-nanodegree-issues #56: Feedback: hope to imporve mentor quality

gwwang16, 2017-06-30:
The Live Chat has a terrible experience besides its interactive interface. I think many mentors are giving misleading suggestions and don't understand concepts correctly. I connected with several mentors before, it turns out all answers I received are wrong.  Moreover, most `mentors` replied very slow, one ridiculous thing is I got nothing useful infomation in one lasted one houre live chat. They are mentor or monster?

lisaljl, 2017-07-03:
My experience is there are a few mentors who actually understands the content and able to explain them. On another note, some just struggles to explain. It is a tough topic, and we definitely need some one that really guide us. Agreed with the slow reply.

gwwang16, 2017-07-04:
Hmm... They must know indeed answers of these common questions now...

kylesf, 2017-07-25:
We are actively assessing the quality of our mentors based on feedback and strive to only provide the best experience. Rest assured it is an important matter for us and hope your future experiences are better!

closed 2017-07-25 by kylesf