udacity/robotics-nanodegree-issues #61: Video tutorials of working with Kinematics Project

fredrick-dominy, 2017-06-30:


fredrick-dominy, 2017-06-30:
One hurdle that I've had is that I don't know what the workflow (editing and testing code) "should" look like in this project. 

There were several hours where I wasn't sure the project was working and only found out later that I have to explicitly call the IK_server.py from a new terminal view.

I'd like to propose a casual video series where a professional starts with our code base and begins modifying the project. (Not focusing on giving us the answers but more focus on the process) From start to close, heck even include the process for version control.

 While doing so, the pro can give tips, show how to use the tools in an effective manner and show us what problems to avoid.  

They can also show us how to debug, test variables and other information that people may not already be aware of.

Thanks Udacity!  Keep up the great work!!

fredrick-dominy, 2017-07-02:
The current videos also have the narrator speaking over a static image.  A good example of this is the DH parameter table video in lesson 16.  

It is very confusing to have a large image with many numbers and letters being talked about without having the narrator draw our attention to the specific letter-combination.  The opposite of this style is used by Khan academy where the narrator will circle, highlight or scribble around areas that he is talking about. 

Another great example is this video on DH parameters.  https://www.youtube.com/watch?v=KstFDEUWUm4    Notice that as the author talks, his meaning is emphasized by drawing your attention to the relevant information. 

Many of the videos Udacity has created for Robotics have had great production quality but they have been woefully inadequate for my needs.  

This has forced me to find other sources  of information and caused me to question whether this paid content is worthwhile.

kylesf, 2017-07-25:
Thanks, for your feedback. We will have a walkthrough video released today hopefully that will help guide a better understanding of the project! We will also make sure we address these concerns in future videos!

closed 2017-07-25 by kylesf