udacity/robotics-nanodegree-issues #65: More guidance on the projects

frozenamazon, 2017-07-03:
It feels like a lot of knowledge has tried to be passed on without much explanation. Yes it is good for the students to self learn, but it is literally throwing into the dark. In the event the material is not updated, more guidance is required. ie mentors that can actually explain and understand the problems faced

sahiljuneja, 2017-07-05:
Is this specific to any project? 

Any particular pain points you faced for any project in regards to this so that we can try focus on that?

kylesf, 2017-07-25:
Updates to this @frozenamazon ?

closed 2017-07-28 by kylesf

lisaljl, 2017-07-28:
I was referring to the kinematics project. But I see improvement on the perception project. Sorry didn't see the notifications , everything went to spam.

kylesf, 2017-07-28:
Well Kinematics has been updated and additional content added!