udacity/robotics-nanodegree-issues #74: error when running safe_spawner.sh

gpavlov2016, 2017-07-09:
[ERROR] [1499639301.473394408, 1113.672000000]: Failed to fetch current robot state
[trajectory_sampler-1] process has died [pid 24948, exit code -11, cmd /home/robond/catkin_ws/devel/lib/kuka_arm/trajectory_sampler __name:=trajectory_sampler __log:=/home/robond/.ros/log/0948f392-64e9-11e7-9385-000c292fac5a/trajectory_sampler-1.log].
log file: /home/robond/.ros/log/0948f392-64e9-11e7-9385-000c292fac5a/trajectory_sampler-1*.log

kylesf, 2017-07-28:
Is this error still and issue for you?

gpavlov2016, 2017-07-31:
With the new VM (2.1.0) I have another problem blocking me from reproducing the original issue - the RVIZ splash screen is stuck at "Initializing" when calling roslaunch kr210_claw_moveit demo.launch but when running rosrun rviz rviz the application opens immediately
[image]

kylesf, 2017-07-31:
Does running the project demo work?

gpavlov2016, 2017-07-31:
No, the problem I described is from running the project demo

kylesf, 2017-08-28:
What resources have you allocated to the VM?

closed 2017-09-05 by kylesf