udacity/robotics-nanodegree-issues #75: Forward and Inverse Kinematics, Lesson 8

sychaichangkun, 2017-07-11:
The solution of Lesson8 Euler Angles from a Rotation Matrix, in Forward and Inverse Kinematics, is obviously wrong.

sahiljuneja, 2017-07-11:
Hey, could you provide more details on what you think is wrong so that we can look into it accordingly?

sychaichangkun, 2017-07-12:
The solution is like this:

beta  = atan2(-r31, sqrt(r11 * r11 + r21 * r21)) * rtd
gamma = atan2(r32, r33) * rtd
alpha = atan2(r21, r11) * rtd
print("alpha is = ",alpha, "radians", "or ", alpha*rtd, "degrees")
print("beta  is = ",beta,  "radians", "or ", beta*rtd, "degrees")
print("gamma is = ",gamma, "radians", "or ", gamma*rtd, "degrees")
-----------------------output---------------------------
alpha is =  45.0000000000000 radians or  2578.31007808870 degrees
beta  is =  60.0000000000000 radians or  3437.74677078494 degrees
gamma is =  30.0000000000000 radians or  1718.87338539247 degrees
-------------------------------------------------------------
alpha, beta and gamma have already been converted to degrees, why muptiply by rtd again?

sahiljuneja, 2017-07-12:
Aah, good catch. Thanks! Fixed.

closed 2017-07-12 by sahiljuneja