udacity/robotics-nanodegree-issues #79: Kinematics - Inverse Kinematics Example typo error

sachin3, 2017-07-19:
This sentence seems to mention x-z plane instead of y-z place. 
"To solve for 𝜃2, imagine 𝜃1 = 0 and project links 2 and 3 onto the the x-z plane. Again, we use the atan2 function,"
Diagram at the top also clearly shows link projected to y-z plane.  x-z plane seems to be typo error.

sahiljuneja, 2017-07-25:
That's not a typo. That's correct. Setting theta 1 to 0 would mean the x-z plane. For y-z plane, theta 1 would have to be 90.

closed 2017-07-25 by sahiljuneja

sachin3, 2017-07-26:
Yes in terms of theta1 = 0 (didn't pay attention to that), it is x-z plane. But I was referring to diagram and theory on the same page, where it seems calculations are done on y-z plane. After discussing this issue with Live Support and making sure my understanding was correct,  I have reported the issue here.  To me also seems diagram and this sentence is not in sync.