udacity/robotics-nanodegree-issues #86: Project Perception - Launch Demo code failed to load (VM 2.1.0)

yannrs, 2017-07-30:
Unable to download model[model://dropbox/meshes/dropbox.dae]

Full Error log:
 /pr2/right_arm_controller/action_monitor_rate: 30
 * /pr2/right_arm_controller/constraints/goal_time: 10.0
 * /pr2/right_arm_controller/gains/right_elbow_flex_joint/d: 1
 * /pr2/right_arm_controller/gains/right_elbow_flex_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_elbow_flex_joint/p: 100
 * /pr2/right_arm_controller/gains/right_forearm_roll_joint/d: 1
 * /pr2/right_arm_controller/gains/right_forearm_roll_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_forearm_roll_joint/p: 100
 * /pr2/right_arm_controller/gains/right_shoulder_lift_joint/d: 1
 * /pr2/right_arm_controller/gains/right_shoulder_lift_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_shoulder_lift_joint/p: 100
 * /pr2/right_arm_controller/gains/right_shoulder_pan_joint/d: 1
 * /pr2/right_arm_controller/gains/right_shoulder_pan_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_shoulder_pan_joint/p: 100
 * /pr2/right_arm_controller/gains/right_upper_arm_roll_joint/d: 1
 * /pr2/right_arm_controller/gains/right_upper_arm_roll_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_upper_arm_roll_joint/p: 100
 * /pr2/right_arm_controller/gains/right_wrist_flex_joint/d: 1
 * /pr2/right_arm_controller/gains/right_wrist_flex_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_wrist_flex_joint/p: 100
 * /pr2/right_arm_controller/gains/right_wrist_roll_joint/d: 1
 * /pr2/right_arm_controller/gains/right_wrist_roll_joint/i: 0.01
 * /pr2/right_arm_controller/gains/right_wrist_roll_joint/p: 100
 * /pr2/right_arm_controller/joints: ['right_shoulder_...
 * /pr2/right_arm_controller/state_publish_rate: 50
 * /pr2/right_arm_controller/stop_trajectory_duration: 0
 * /pr2/right_arm_controller/type: position_controll...
 * /pr2/right_gripper_controller/constraints/goal_time: 3.0
 * /pr2/right_gripper_controller/constraints/right_left_gripper_finger_joint/goal: 0.02
 * /pr2/right_gripper_controller/constraints/right_right_gripper_finger_joint/goal: 0.02
 * /pr2/right_gripper_controller/gains/right_left_gripper_finger_joint/d: 10
 * /pr2/right_gripper_controller/gains/right_left_gripper_finger_joint/i: 1
 * /pr2/right_gripper_controller/gains/right_left_gripper_finger_joint/i_clamp: 1.0
 * /pr2/right_gripper_controller/gains/right_left_gripper_finger_joint/p: 100
 * /pr2/right_gripper_controller/gains/right_right_gripper_finger_joint/d: 10
 * /pr2/right_gripper_controller/gains/right_right_gripper_finger_joint/i: 1
 * /pr2/right_gripper_controller/gains/right_right_gripper_finger_joint/i_clamp: 1.0
 * /pr2/right_gripper_controller/gains/right_right_gripper_finger_joint/p: 100
 * /pr2/right_gripper_controller/joints: ['right_right_gri...
 * /pr2/right_gripper_controller/type: effort_controller...
 * /pr2/torso_lift_joint_controller/joint: torso_lift_joint
 * /pr2/torso_lift_joint_controller/pid/d: 1.0
 * /pr2/torso_lift_joint_controller/pid/i: 0.1
 * /pr2/torso_lift_joint_controller/pid/p: 100.0
 * /pr2/torso_lift_joint_controller/type: position_controll...
 * /pr2/world_joint_controller/joint: world_joint
 * /pr2/world_joint_controller/pid/d: 1.0
 * /pr2/world_joint_controller/pid/i: 0.1
 * /pr2/world_joint_controller/pid/p: 100.0
 * /pr2/world_joint_controller/type: position_controll...
 * /pr2_cloud_transformer/demo: True
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/left_arm/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/left_arm/kinematics_solver_attempts: 3
 * /robot_description_kinematics/left_arm/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/left_arm/kinematics_solver_timeout: 0.005
 * /robot_description_kinematics/right_arm/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/right_arm/kinematics_solver_attempts: 3
 * /robot_description_kinematics/right_arm/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/right_arm/kinematics_solver_timeout: 0.005
 * /robot_description_planning/joint_limits/left_elbow_flex_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/left_elbow_flex_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/left_elbow_flex_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/left_elbow_flex_joint/max_velocity: 3.3
 * /robot_description_planning/joint_limits/left_forearm_roll_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/left_forearm_roll_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/left_forearm_roll_joint/max_acceleration: 0
[... 379 lines cut ...]
[move_group-5] killing on exit
[pcl_manager-6] killing on exit
[joint_state_relay-3] killing on exit
[robot_state_publisher-2] killing on exit
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to connect to move_group action server 'move_group' within allotted time (0s)
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

[gazebo-9] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

kylesf, 2017-07-31:
What are the resources that you are currently providing your VM?

yannrs, 2017-07-31:
I tried several times (>5 times) with 4 cores, 2Go of Ram and 512Mo GPU, and with 6cores. (i7-3630QM 2.4GHz to 3.4GHz), while only 600Mo of Ram is used.
I also tried with the VM 2.0.1 with the same results.

yannrs, 2017-07-31:
I also tried with the last version of the github (today) with the same results.

kylesf, 2017-07-31:
A solution is in the works. Will keep you updated!

kylesf, 2017-07-31:
The project repo has been updated to fix this issue!

closed 2017-07-31 by kylesf

yannrs, 2017-08-01:
Thanks ! Looks better