udacity/robotics-nanodegree-issues #89: Error initializing Outlier Removal Filter using  make_statistical_outlier_filter()

rdspring1, 2017-08-02:


rdspring1, 2017-08-02:
There is a mistake in the python-pcl build. The constructor for the filter requires the input point cloud as an argument, but the python interface is not passing the argument.

Here is my fix:
Remove the unnecessary argument. Reinstall python-pcl.
Part 1:
In pcl/pxi/Filters/StatisticalOutlierRemovalFilter_172.pxi

Change 
```
def __cinit__(self, PointCloud pc not None):
         self.me = new pclfil.StatisticalOutlierRemoval_t()
         (<cpp.PCLBase_t*>self.me).setInputCloud (pc.thisptr_shared)
```
to
```
def __cinit__(self):
         self.me = new pclfil.StatisticalOutlierRemoval_t()
```
Repeat for all instance of the constructor in file

Part2:
In pcl/pxi/PointCloud_PointXYZ_172.pxi
Change
```
def make_statistical_outlier_filter(self):
    # fil = StatisticalOutlierRemovalFilter()
    # cdef pclfil.StatisticalOutlierRemoval_t *cfil = <pclfil.StatisticalOutlierRemoval_t *>fil.me
    # cfil.setInputCloud(<cpp.shared_ptr[cpp.PointCloud[cpp.PointXYZ]]> self.thisptr_shared)
    return StatisticalOutlierRemovalFilter(self)
```
to
```
def make_statistical_outlier_filter(self):
      fil = StatisticalOutlierRemovalFilter()
      cdef pclfil.StatisticalOutlierRemoval_t *cfil = <pclfil.StatisticalOutlierRemoval_t *>fil.me
      cfil.setInputCloud(<cpp.shared_ptr[cpp.PointCloud[cpp.PointXYZ]]> self.thisptr_shared)
      return fil
```

Part 3:
1) Delete pcl/_pcl_172.cpp
It is auto-generated by cython. The build process skips this step if it is already present

Part 4:
Reinstall python-pcl

ryan-keenan, 2017-08-03:
Thanks for debugging this!  I've tested and confirmed that this fix works and pushed the change to the RoboND-Perception-Exercises repo.

closed 2017-08-03 by ryan-keenan