udacity/robotics-nanodegree-issues #90: Lesson5, exercise3.A big cude in Gazebo when capturing features

sychaichangkun, 2017-08-03:
When capturing features in exercise 3, sometimes a big cube would appear and stay still, it's not supposed to be in the model list.

ryan-keenan, 2017-08-03:
Thanks for reporting this!  I haven't seen that behavior myself.  Are you using the VM or local Linux/ROS install?

sychaichangkun, 2017-08-05:
I'm using local Linux+ROS, but a student from live help reported that issue using VM.

sahiljuneja, 2017-08-12:
Is this still an issue? Because we have been unable to reproduce this on the VM as of now to be able to pinpoint what could be causing it.

jchernus, 2017-08-14:
I experience this too.  Running in VM.

yuan0623, 2017-08-28:
I'm experiencing the big cube, what's that?

kylesf, 2017-08-28:
@hvpandya Any idea what the cube may be?

hvpandya, 2017-09-05:
The only two reasons for a cube to appear instead of a given model are:
1) Incorrect GAZEBO_MODEL_PATH in your .bashrc
2) The model file/s are not present at the location you defined as GAZEBO_MODEL_PATH

kylesf, 2017-09-05:
Thanks, Harsh. We will now close this issue.

closed 2017-09-05 by kylesf