udacity/robotics-nanodegree-issues #91: Feedback - Github commit messages lack detail

adammalpass, 2017-08-04:
Hi team,

Firstly thanks for making quick changes to the project files in GitHub when we find some bugs. But just one request, please can you add more detailed commit messages so it is easy to tell what has been changed? For example a recent commit to the Perception project simply said "Update project_template.py". It's hard to tell if this is a large critical update that needs integrating into our project, or just something minor we can ignore. And if many changes how we should consider including merging them into our design, etc.

Udacity has its own GitHub commit style guide, so please can I suggest it is used for this course too? It would make our lives much easier ;)

https://udacity.github.io/git-styleguide/

adammalpass, 2017-08-04:
*Also it would set a good role-model for other students how to use GitHub properly! Thanks!

ryan-keenan, 2017-08-04:
Great feedback Adam!  We will try to write better commit messages going forward ;)

closed 2017-08-04 by ryan-keenan