udacity/robotics-nanodegree-issues #92: Lesson 6 - Perception project - I don't see the gazebo world.

Things2Connect, 2017-08-04:
In lesson 6.4 "Project Demo" .- Once Gazebo is up and running, I do not see following objects in the Gazebo world:

    Robot
    Table arrangement
    Three target objects on the table
    Dropboxes on either sides of the robot

Instead, I just see a simplifed picture of the PR2 with no detailed objects.

As instructed, I have removed the 'gazebo_grasp_plugin' directory from the `RoboND-Perception-Project/` directory, but this does not seem to help.

I get this error message in the terminal:

[ERROR] [1501843718.270276770, 992.297000000]: "world" passed to lookupTransform argument target_frame does not exist. 

(My configuration : VM = robond_5_22_17, 4 GB RAM, 4 Cores, Intel i7-7500U)

Things2Connect, 2017-08-04:
I have attached snapshot from Gazebo:

[image]

sahiljuneja, 2017-08-04:
Hey. This usually happens when your GAZEBO_MODEL_PATH is not properly set in `.bashrc`. I suggest you double-check that.

Things2Connect, 2017-08-04:
Thanks.  I got the project demo to work now :-)

I double-checked my .bashrc file and have made a few corrections. The objects load fine now (robot + table arrangement + target boxes + dropboxes).

a) I had placed line:

"source ~/catkin_ws/devel/setup.bash" 

before line:

 "export GAZEBO_MODEL_PATH=~/catkin_ws/src/RoboND-Perception-Project/pr2_robot/models" 

I have rearranged these in opposite order.

b) Also,  from lesson 6.4, I removed the last bit ("_:$GAZEBO_MODEL_PATH_") from line:

export GAZEBO_MODEL_PATH=~/catkin_ws/src/RoboND-Perception-Project/pr2_robot/models**:$GAZEBO_MODEL_PATH**

As it seemed unnecessary.

So, the demo works now :-)

sahiljuneja, 2017-08-04:
That's awesome! Thanks for confirming. Good luck with the project! :)

closed 2017-08-04 by sahiljuneja