udacity/robotics-nanodegree-issues #95: Video and theory defining D-H parameters are not in sync

sachin3, 2017-08-10:
In theory, Xi is defined as per Z(i) and Z(i+1) but while explaining the same concept, instructor mentions it Z(i) and Z(i-1), which is incorrect. A diagram which shows X2 in the same video is drawn as per the explanation is given in theory. 
Video link - (around 1:15 timeline) https://classroom.udacity.com/nanodegrees/nd209/parts/7b2fd2d7-e181-401e-977a-6158c77bf816/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/b1673850-9441-41b8-ae3a-5578285067b2
Theory - https://classroom.udacity.com/nanodegrees/nd209/parts/7b2fd2d7-e181-401e-977a-6158c77bf816/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/2f59c902-9c32-4b26-9e52-5e495ec14dba

sahiljuneja, 2017-08-10:
Hey, I believe this is in relation to the discussion you were having with another student in the Slack?

If so, then for calculating X_2 you have Z_2 and Z_3 as parallel axis and therefore the assignment is arbitrary as has been mentioned in the classroom (although, in many cases, it is also relatively simple to have it assigned as the common normal between the two)

I believe you also came to the same conclusion from the Slack, so I will close this for now :)

sachin3, 2017-08-10:
Hi.
Yes, that discussion was revolving around this point.
But this issue is different... have you checked video where the narrator was mentioning Z(i) and Z(i-1) instead of Z(i) and Z(i+1)? Is that correct?

sahiljuneja, 2017-08-10:
Yes, I am looking into that now. It is still correct in terms of having it defined in that way as the common normal between the two, but I am looking into consistency on the phrasings to be less confusing. Thanks for pointing this out. Opening this ticket for now.

sachin3, 2017-08-10:
Oops, I logged another issue as there was no way to open this one again. 

Yes, diagram and earlier theory in the lesson are in sync but just video caption.

closed 2017-10-05 by sahiljuneja