udacity/robotics-nanodegree-issues #96: Video and theory about D-H x parameter explanation is not in sync

sachin3, 2017-08-10:
Sorry for logging this issue again. I feel the actual issue is not addressed yet.

This video - 
https://classroom.udacity.com/nanodegrees/nd209/parts/7b2fd2d7-e181-401e-977a-6158c77bf816/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/2f59c902-9c32-4b26-9e52-5e495ec14dba

Timeline - 1:32 -> The X-axes define the common normals between Z_i-1 and Z_i. Check out the closed caption on video.

Shouldn't this be Z_i and Z_i+1

sachin3, 2017-08-10:
Correct video link referred above - https://classroom.udacity.com/nanodegrees/nd209/parts/7b2fd2d7-e181-401e-977a-6158c77bf816/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/91d017b1-4493-4522-ad52-04a74a01094c/concepts/398f60b1-b112-41ec-b8bc-59e95a2cec70

sahiljuneja, 2017-08-10:
I have already reopened your previous issue and pointed that I am looking into it. I will close this one to avoid duplicates.  Thanks!

closed 2017-08-10 by sahiljuneja