udacity/robotics-nanodegree-issues #97: Typos Perception Project

mi7flat5, 2017-08-10:
In section 3 'Environment Setup' 'export GAZEBO_MODEL_PATH=~/catkin_ws/src/RoboND-Perception-Project/pr2_robot/models:$GAZEBO_MODEL_PATH
'
does not work for me, I had to add brackets to GAZEBO_MODEL_PATH

export GAZEBO_MODEL_PATH=~/catkin_ws/src/RoboND-Perception-Project/pr2_robot/models:${GAZEBO_MODEL_PATH}

In section 6 'Output yaml file' The command '$ rosmsg info geomtery_msgs/Pose' geometry is misspelled.

sahiljuneja, 2017-08-10:
Fixed the typo, thanks!

As for the brackets, that hasn't been the case for anyone till now. Are you working on the VM? Not sure why that would happen. What exactly do you face if you don't use brackets?

mi7flat5, 2017-08-10:
I'm on a local install of ubuntu. I've had that line work in the VM and another local install but not this one, it fails to find the models. After doing some google searching I have found that the brackets are common and it fixed my issue, so perhaps it is a more general approach? If not disregard. for reference http://answers.gazebosim.org/question/1940/problem-with-including-a-model/

sahiljuneja, 2017-09-13:
Haven't had any similar problems from students about this specific to the VM, so I will close this for now.

closed 2017-09-13 by sahiljuneja