udacity/robotics-nanodegree-issues #102: Controls lesson 1.5

dandrews7396, 2017-08-17:
quiz asks for the minimum control effort for the quadrotor to lift, to 5s.f. It accepts the answer 1.5107, which is taken from the result of g*m/c (1.51074). However, as is it even stated within the code, the quadrotor will not take off until the control force is greater than g*m/c. This means that the required answer of 1.5107 is incorrect and, as the actual required force is 1.51075, should be 1.5108.

kylesf, 2017-08-17:
Fixed, thanks!

closed 2017-08-17 by kylesf

gurparteek, 2017-08-29:
1.510741 is also greater than 1.51074, so is 1.5107400000001, the greater number need not necessarily be 1.51075. The original answer of 1.5107 was correct.

kylesf, 2017-08-29:
Okay fair point, I will change back

dandrews7396, 2017-08-29:
No it's not. I'm currently mobile, but if I remember correctly, the code
contains the conditional statement that it must be greater than, not equal
to. If it is equal to, then that condition will not be met and it will
fail. Maybe my example was flawed, but my statement still stands

kylesf, 2017-08-29:
Made it 3 significant figures with 1.52 as answer.