udacity/robotics-nanodegree-issues #104: Issue with Launch File for Write ROS Nodes 3-10 'Look away'

adammalpass, 2017-08-19:
I had an issue running the look_away code in the Write ROS nodes section 3-10. Finally I managed to trace the issue back to the way the launch file is set up in section 3-10. There can exist a race condition whereby the look_away node starts up before the arm_mover node is ready and hence the look_away node's look_away_callback function gets permanently stuck at the 'rospy.wait_for_service('/arm_mover/safe_move')' line.

This can be fixed by creating a safe_spawner type script (like we do for later modules) or more easily just leaving the original launch file untouched and then instructing students to start the look_away node manually.

This was really complicated to debug, so please let everyone know to save lots of hassle later! :)

adammalpass, 2017-08-19:
Btw I created and tested a safe_spawner.sh script to fix this issue. Now the project runs with no issues. I created a pull request on the simple arm repo to add this script, so if you like please accept so other students can share it :)

kylesf, 2017-08-28:
I will check this out. Thanks :)

adammalpass, 2017-08-31:
Hi Kyle I had a email saying the pull request was good and accepted, but it doesn't seem to have made it into the repo yet for some reason. I'm not expert in pull requests, so please could you check? Thanks! :)

Deborah-Digges, 2017-09-02:
I spent about an hour wondering why `look_away` wasn't working! this turned out to be the issue. Quite frustrating really. Please could the lessons be updated to at least reflect the explanation given by @adammalpass

adammalpass, 2017-09-02:
@kylesf Thanks for accepting the pull request to add my safe spawner script. Perhaps you could also update the lesson notes to recommend using it to, to help other students avoid finding this issue? Thanks!

adammalpass, 2017-09-02:
@Deborah-Digges Glad it fixed your issue! Hopefully it can get added to lesson notes soon :)

kylesf, 2017-09-07:
Done!

closed 2017-09-07 by kylesf