udacity/robotics-nanodegree-issues #109: controls lesson 2-5

jupidity, 2017-08-25:


jupidity, 2017-08-25:
I'm nitpicking a bit here, but on the question regarding what parameters need to be zero for a quadcopter to hover, the answer is only true if the force vector points along the negative z axis, which isnt true in the presence of wind or other external forces. 

For example if the net force vector on the copter pointed along either the x or y axis slightly, you would need to counteract that with non-zero pitch and yaw angels. 

I think the answer should be only the velocity vectors need to be zero, since that guarantees the forces are balanced.

Also, in the presence of ideal conditions (force vector along the negative z-axis), dpsi/dt could be nonzero and the copter could hover. 

Like I said, nit-picky, but maybe it matters.

jupidity, 2017-08-25:
Actually, I just realized that it's probably in the context of determining the zero parameters for the hover controller in the Unity sim, which has ideal conditions. So maybe better to leave it. 

But it is paired with an overview of statics, which essentially says the rpy angles could be non-zero while hovering in the presence of external forces besides gravity, so maybe its still worth addressing.

kylesf, 2017-08-28:
We will keep this in mind. Thanks for the feedback!

thomas-at-em, 2017-09-06:
Thanks for the feedback. As you can see, it is a challenge to write questions that emphasize the intended concept and also avoid being overly pedantic. As you can see from the derivation of the nominal motor speed at an equilibrium hover, all four motors have the same angular speed, thus there could be no rotation about the bz-axis.  But I will add a note explicitly stating that all external disturbances are assumed zero.

closed 2017-09-06 by thomas-at-em