udacity/robotics-nanodegree-issues #112: Fix training_helper.py glitches/crashing

croomjm, 2017-08-27:
Master version of src/sensor_stick/training_helper.py has the following issues:
1) Object may not always be deleted between object types, so the new model may not get loaded.
2) Occasional crashes due to attempting to spawn new model with the same name before the old model has been deleted.
3) If the script is stopped with an interrupt, gazebo has to be rebooted to start capturing again since the script does not attempt to delete any remaining objects in the scene before starting.

I added the following lines before the assignment of 'spawn_model_prox", which seemed to solve most of the issues above:
    try:
        delete_model()
    except:
        print('No model to delete.')

kylesf, 2017-08-28:
@hvpandya Thoughts on this?

hvpandya, 2017-09-05:
I believe this script was written by @bkinman

bkinman, 2017-09-08:
By adding more service calls we can fix all of the problems mentioned above. However, this would reduce the overall speed. One great way to fix this is by writing a gazebo plugin to assist us with the process of gathering training data.

It's not a bad idea to add the lines above... But I think that we need to write a gazebo plugin at some point.

closed 2018-05-01 by jessluls