udacity/robotics-nanodegree-issues #113: Perception objects levitate away

croomjm, 2017-08-27:
Due to lack of friction and low mass, some of the objects in both the perception exercise 3 and project (especially the create) slowly glide away from initial state even with no applied force. Objects should be updated with appropriate friction/mass. Objects in perception exercise 3 could be fixed in space since they don't need to be grabbed.

kylesf, 2017-08-28:
@hvpandya Thoughts on this?

hvpandya, 2017-09-05:
The create does not glide because of lack of friction, it has a motion controller embedded in it since its motion is intentional (consider it an easter egg). 

I do agree that at some point we should update all models with realistic mass, friction and damping properties but that is not something we will pursue in the near future. @kylesf Feel free to keep this ticket in the backlog.

closed 2018-05-01 by jessluls