udacity/robotics-nanodegree-issues #121: Controls 1-22 Putting it all together - code outline is dangerously wrong!

adammalpass, 2017-08-31:
In controls 1-22 there is a code outline for PID controller, but if you follow the outline it will _not_ work! Look at this section:

```# Sum the errors
        self.error_sum_ += error * delta_time
        
        # Update the past error
        self.last_error_ = error
        
        # Find delta_error
        delta_error = error - self.last_error_
        
        # Update the past error
        self.last_error_ = error```

Firstly notice there is an instruction to set self.last_error_ twice, once before and once after delta_error. The first one _must_ be removed. In the current code `delta_error` will always be 0 and hence the derivative term has no effect.

I removed this one line from my code and now everything works perfectly

adammalpass, 2017-08-31:
Also the solution.py on same page contains same issue - please fix too!

kylesf, 2017-08-31:
Fixed, thanks adam!

closed 2017-08-31 by kylesf