udacity/robotics-nanodegree-issues #122: Controls 2.12 - rqt reconfigure fails

Things2Connect, 2017-09-01:
Controls, lesson 2-12, "dynamic_reconfigure" - I cannot launch rqt_reconfigure with command: "$ rosrun rqt_reconfigure rqt_reconfigure"

These steps work fine:
a) Launch of roscore
b) Launch of drone simulator
c) Checking of topic /quad_rotor/pose for messages
d) Launch of hover controller node

However, when I try to launch the rqt_reconfigure instance, I get following error:

$ rosrun rqt_reconfigure rqt_reconfigure
[rospack] Error: package 'rqt_reconfigure' not found

Somehow the package is missing on my VM ?
(I'm using the lastest VM version: V2.1.0)

Things2Connect, 2017-09-01:
The rqt_reconfigure package is present on my 'old' VM (5.22.17) but, somehow, it is not present on my new VM (2.1.0):

**VM (2.1.0) :**
robond@udacity:~$ rospack find rqt_reconfigure
[rospack] Error: package 'rqt_reconfigure' not found
robond@udacity:~$ 

**VM 5.22.17**
robond@udacity:~$ rospack find rqt_reconfigure
/opt/ros/kinetic/share/rqt_reconfigure

- I have a re-run Software Updater inside my new VM (2.1.0), and it says "The software .....is up-to-date", so there are no updates left to download.

Things2Connect, 2017-09-01:
The problem has now been solved!   I ran the apt-get command below in the terminal instead of using menu option System Tools/ Software Updater in the VM:

sudo apt-get install ros-kinetic-rqt-reconfigure
 
So, rqt_reconfigure works now.

kylesf, 2017-09-01:
Cool, It seems this is installed in the current VM. Not sure what may have happened.

closed 2017-09-01 by kylesf