udacity/robotics-nanodegree-issues #123: Controls - Lesson 2.14 - Positional Controller

Things2Connect, 2017-09-02:
I get an error when I carry out a roslaunch for the positional controller in lesson 2.14:                                                        roslaunch quad_controller position_controller.launch                                     I don't know if it is related to tickets #114 and #115.  If I instead do an individual rosrun on the positional controller node , it works fine.  There seems to be some kind issue in the launch file for the positional controller.

Things2Connect, 2017-09-02:
Here is the terminal output:                                                                                                robond@udacity:~$ roslaunch quad_controller position_controller.launch
... logging to /home/robond/.ros/log/878fd22a-8fd2-11e7-a2e7-000c29100b05/roslaunch-udacity-16151.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.19.129:46513/

SUMMARY
========

PARAMETERS
 * /attitude_controller/pitch_kd: 0.0
 * /attitude_controller/pitch_ki: 0.0
 * /attitude_controller/pitch_ki_max: 0.0
 * /attitude_controller/pitch_kp: 0.0
 * /attitude_controller/roll_kd: 0.0
 * /attitude_controller/roll_ki: 0.0
 * /attitude_controller/roll_ki_max: 0.0
 * /attitude_controller/roll_kp: 0.0
 * /attitude_controller/yaw_kd: 0.0
 * /attitude_controller/yaw_ki: 0.0
 * /attitude_controller/yaw_ki_max: 0.0
 * /attitude_controller/yaw_kp: 0.0
 * /position_controller/x_kd: 0.0
 * /position_controller/x_ki: 0.0
 * /position_controller/x_ki_max: 0.0
 * /position_controller/x_kp: 0.0
 * /position_controller/y_kd: 0.0
 * /position_controller/y_ki: 0.0
 * /position_controller/y_ki_max: 0.0
 * /position_controller/y_kp: 0.0
 * /position_controller/z_kd: 0.0
 * /position_controller/z_ki: 0.0
 * /position_controller/z_ki_max: 10
 * /position_controller/z_kp: 0.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    attitude_controller (quad_controller/attitude_controller_node)
    position_controller (quad_controller/position_controller_node)
    quaternion_to_euler_node (quad_controller/quaternion_to_euler_node)
    rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[position_controller-1]: started with pid [16170]
process[attitude_controller-2]: started with pid [16171]
process[quaternion_to_euler_node-3]: started with pid [16172]
process[rqt_reconfigure-4]: started with pid [16173]
[INFO] [1504352060.975659]: Attitude controller config changed!
[ERROR] [1504352061.129975]: bad callback: <bound method PositionControllerNode.pose_callback of <__main__.PositionControllerNode instance at 0x7f2de875fe18>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node", line 116, in pose_callback
    if self.counter >= 10 or not self.first_pose_received_:
AttributeError: PositionControllerNode instance has no attribute 'counter'

[INFO] [1504352061.136584]: Position controller node config changed!
[ERROR] [1504352061.158853]: bad callback: <bound method PositionControllerNode.pose_callback of <__main__.PositionControllerNode instance at 0x7f2de875fe18>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node", line 116, in pose_callback
    if self.counter >= 10 or not self.first_pose_received_:
AttributeError: PositionControllerNode instance has no attribute 'counter'

[ERROR] [1504352061.174335]: bad callback: <bound method PositionControllerNode.pose_callback of <__main__.PositionControllerNode instance at 0x7f2de875fe18>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node", line 116, in pose_callback
    if self.counter >= 10 or not self.first_pose_received_:
AttributeError: PositionControllerNode instance has no attribute 'counter'

Things2Connect, 2017-09-02:
If instead, I restart the system, and I carry out indiviudal an rosrun on each components in the launch file of the position controller:  robond@udacity:~$ rosrun quad_controller position_controller_node
[INFO] [1504353054.984205]: Position controller node config changed!

Things2Connect, 2017-09-02:
I've managed to solve the issue now with help from others in the slack channel.

kylesf, 2017-09-05:
What is the current status of this issue?

Things2Connect, 2017-09-06:
I tried to reboot the VM to remove the errror (as recommended on Slack channel for Controls).  Also, I ran a new catkin_make.  However, these actions did not remove the error in the end.

Instead, I managed to remove the error by modifying the script "position_controller_node", as below:

--> moved the line  "self.counter = 0" to the beginning of the constructor belonging to class PositionControllerNode()

See code snippet below.

I'm not sure exactly why, but once I modified the constructor as mentioned, the position controller worked fine.  

I have now managed to complete the lab for Controls.   My drone can fly out to the lava cube, land and return back.  

---------------------------------

class PositionControllerNode():
    def __init__(self):
        self.first_pose_received_ = False
        self.counter = 0  #moved self.counter to beginning of the constructor
        # X PID Config
        x_kp = float(rospy.get_param('x_kp', '0'))
        x_ki = float(rospy.get_param('x_ki', '0'))
        x_ki_max = float(rospy.get_param('x_ki_max', '0'))
        x_kd = float(rospy.get_param('x_kd', '0'))
        self.x_controller_ = PIDController(x_kp, x_ki, x_kd, x_ki_max)

        # Y PID Config
        y_kp = float(rospy.get_param('y_kp', '0'))
        y_ki = float(rospy.get_param('y_ki', '0'))
        y_ki_max = float(rospy.get_param('y_ki_max', '0'))
        y_kd = float(rospy.get_param('y_kd', '0'))
        self.y_controller_ = PIDController(y_kp, y_ki, y_kd, y_ki_max)

        # Z PID Config
        z_kp = float(rospy.get_param('z_kp', '0'))
        z_ki = float(rospy.get_param('z_ki', '0'))
        z_ki_max = float(rospy.get_param('z_ki_max', '0'))
        z_kd = float(rospy.get_param('z_kd', '0'))
        self.z_controller_ = PIDController(z_kp, z_ki, z_kd, z_ki_max)

        self.pose_sub_ = rospy.Subscriber('/quad_rotor/pose', PoseStamped, self.pose_callback)

        self.goal_point_sub_ = rospy.Subscriber('/quad_rotor/goal_point', Point, self.goal_point_callback)
        self.reset_srv_ = rospy.Service('/position_controller/controller_reset', Empty,  
        self.controller_reset_callback)

        self.cmd_z_thrust_pub_ = rospy.Publisher('/quad_rotor/cmd_z_thrust', Float64, queue_size=10)
        self.cmd_attitude_pub_ = rospy.Publisher('/quad_rotor/cmd_attitude', Vector3, queue_size=10)

        srv = Server(position_controller_paramsConfig, self.config_callback)
        #self.counter = 0   #Original location of self.counter
        .
        .
        .

kylesf, 2017-09-07:
Okay we will take a look at it!

bkinman, 2017-09-07:
Excellent find! The real problem here is that we are calling init_node before we initialized the `PositionControllerNode` class.

Because of this, the pose callback (which depends upon the counter) is getting called before the counter has been declared (aka PositionControllerNode's __init__() function is still running).

The fix that you've implemented will probably work, but we should instead be calling init_node() after PositionControllerNode has been initialize.d

kylesf, 2017-09-08:
Made the changes. Can you confirm it working now @Things2Connect

Things2Connect, 2017-09-09:
I've done a git pull tonight to download the above-mentioned changes onto my local repository.  However, I get an error when I re-run a roslaunch for the position controller node.  The systems seems to have a problem creating topics before doing a node init?  I've attached the terminal output below:

robond@udacity:~$ roslaunch quad_controller position_controller.launch
... logging to /home/robond/.ros/log/187d73c6-957f-11e7-b68b-000c29100b05/roslaunch-udacity-11046.log
.
.
.
.
.
ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[position_controller-1]: started with pid [11065]
process[attitude_controller-2]: started with pid [11066]
process[quaternion_to_euler_node-3]: started with pid [11067]
process[rqt_reconfigure-4]: started with pid [11068]
Traceback (most recent call last):
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node", line 155, in <module>
    ac = PositionControllerNode()
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node", line 70, in __init__
    srv = Server(position_controller_paramsConfig, self.config_callback)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/dynamic_reconfigure/server.py", line 78, in __init__
    self.descr_topic = rospy.Publisher(self.ns + 'parameter_descriptions', ConfigDescrMsg, latch=True, queue_size=10)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 842, in __init__
    super(Publisher, self).__init__(name, data_class, Registration.PUB)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 153, in __init__
    self.resolved_name = rospy.names.resolve_name_without_node_name(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/names.py", line 130, in resolve_name_without_node_name
    raise ValueError("~name topics cannot be created before init_node() has been called")
ValueError: ~name topics cannot be created before init_node() has been called
[position_controller-1] process has died [pid 11065, exit code 1, cmd /home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/position_controller_node __name:=position_controller __log:=/home/robond/.ros/log/187d73c6-957f-11e7-b68b-000c29100b05/position_controller-1.log].
log file: /home/robond/.ros/log/187d73c6-957f-11e7-b68b-000c29100b05/position_controller-1*.log
[INFO] [1504976150.134648]: Attitude controller config changed!
[rqt_reconfigure-4] process has finished cleanly
log file: /home/robond/.ros/log/187d73c6-957f-11e7-b68b-000c29100b05/rqt_reconfigure-4*.log

kylesf, 2017-09-09:
I reverted the changes to your original suggestion. Hopefully this will resolve the issue. Thanks for the follow up

Things2Connect, 2017-09-09:
Thanks.  Yes, the issue is resolved now.  Just tried again. The simulator works fine and can position the drone onto the lava cube.

closed 2018-03-12 by kylesf