udacity/robotics-nanodegree-issues #124: Controls - Lesson 2.12 - hover zn tuner node does not work

Things2Connect, 2017-09-02:
I have problems running the script for the hover_zn_tuner_node.  I get an error saying that I don't have the matplotlib module.  I don't understand this?  I am using the latest VM (V2.1.0).  The script for the twiddle_tuner_node runs fine.

Things2Connect, 2017-09-02:
Here is the output for my terminal:......    robond@udacity:~$ rosrun quad_controller hover_zn_tuner_node
Traceback (most recent call last):
  File "/home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/hover_zn_tuner_node", line 3, in <module>
    import matplotlib.pyplot as plt
ImportError: No module named matplotlib.pyplot
robond@udacity:~$ rosrun quad_controller hover_twiddle_tuner_node
Initial Params: kp:0.0, ki:0.0, kd:0.0
iterations: 1 params: [0.0, 0.0, 0.0] best_run: 11601.7470335
iterations: 2 params: [0, 0.2, 0.2] best_run: 11522.6559698                                                           iterations: 3 params: [0.22000000000000003, 0.19999999999999998, 0.19999999999999998] best_run: 11522.6559698

Things2Connect, 2017-09-02:
(continued)  robond@udacity:~$ rosrun quad_controller hover_twiddle_tuner_node
Initial Params: kp:0.0, ki:0.0, kd:0.0
iterations: 1 params: [0.0, 0.0, 0.0] best_run: 11601.7470335
iterations: 2 params: [0, 0.2, 0.2] best_run: 11522.6559698
iterations: 3 params: [0.22000000000000003, 0.19999999999999998, 0.19999999999999998] best_run: 11522.6559698
iterations: 4 params: [0.22, 0.19999999999999996, 0.19999999999999996] best_run: 11522.6559698

Things2Connect, 2017-09-02:
I have solved this problem now by running following command:

pip install matplotlib

I am not sure why the module was missing on the VM (I am using latest version: V2.1.0).  But it works now, and I able to run the hover_zn_tuner_node.

kylesf, 2017-09-07:
Not sure why that is the case but I installed it just to be safe. Thanks

closed 2017-09-07 by kylesf