udacity/robotics-nanodegree-issues #133: DroneSim sometimes not accepting changes in rqt_reconfigure

TillScout, 2017-09-07:
On some occasions (I have not been able to reproduce) the DroneSim does not react to changes that are done in rqt_reconfigure. That means that e.g. during with the attitude_controller node I tune the parameters for pitch control and after tuning a couple of minutes it does not react to changes in the yaw parameters, but still to changes that I did initially. Restarting roscore, the sim and the node solves the problem, but then in most cases it comes back again with some other parameters that are ignored.
I use the sim on native Ubuntu, but on Slack I heard that this also happens with the Windows version of the sim.

kylesf, 2017-09-07:
Are you using the provided VM for the course?

bkinman, 2017-09-07:
I'm not sure what is causing the problem here.

It's like likely either a problem with networking, or more likely, a problem in ROS.NET/DroneSim.

I myself have not experienced this particular problem. I _have_ experienced rare and intermittent connectivity issues. For me, the problem was solved by restarting roscore/DroneSim

It will be excellent when we get to the root-cause of this problem.

TillScout, 2017-09-08:
I use a native Ubuntu installation. On Slack someone mentioned that the same problem occurs on Windows, so I assume that he used the supplied VM.
The problem makes it barely possible to come up with properly tuned PID parameters. Restarting roscore/DroneSim helps, but after a few seconds the problem comes back in most situations.

TillScout, 2017-09-09:
I noticed that constraining and unconstraining the problematic rotation/position temporarily fixes the problem. But it comes back after a while

yulivee, 2017-09-09:
I too, experience this problem. In my case, the sim never accepts input. I do not even havce phases where it works. Host OS is Ubuntu Linux

TillScout, 2017-09-09:
@yulivee In rqt_reconfigure, did you check the box saying "use_dr_set_point" at the top? It took me several hours to find out that you have to tick that box, otherwise all your changes are ignored.
Might be worth adding that piece of information to the classroom, by the way...

yulivee, 2017-09-10:
@TillScout Where is that checkbox? I cannot find it in rqt_reconfigure

closed 2018-05-01 by jessluls