udacity/robotics-nanodegree-issues #143: Control: Lesson 18. Beyond the Ideal Case - Noise

ancabilloni, 2017-09-23:
The solution of low-pass filter is different from the formula in the video.
On line 26 of the solution, the code is `filtered[i] = filtered[i-1] + alpha*(signal[i] - filtered[i-1])`

However, according to the lecture, it should be ` filtered[i] = filtered[i-1] + alpha*(signal[i] - filtered[i-1])`

There aren't significant differences between two solutions, but should the second one is more a correct one to use in the future?

ancabilloni, 2017-09-23:
This is the formula in the video:

[image]

satsugaisha, 2018-03-02:
I've also come across this problem, but there's one mistake with what @ancabilloni  has written, the actual equation, according to the video in the lesson, should be `alpha * ((signal[i] - signal[i-1])/(t[i] - t[i-1])) + (1 - alpha) * filtered[i-1]` which produces at least as much, if not more, noise than the original signal.