udacity/robotics-nanodegree-issues #158: Please communicate changes to project files

adammalpass, 2017-10-02:
Seems some updates were made to support functions in Deep Learning project over the weekend (for example preprocess_ims.py).

Bug fixes are good, but not communicating changes isn't. I had no idea changes were made and still have no idea what the update is about. Do I need to update? Is all the work I made so far unusable due to whatever bug was found?

Please provide more detailed commit messages and also post on Slack so we know when important changes are made, and what happened.

sahiljuneja, 2017-10-02:
Sometimes there are some important fixes as per some bug reports or some testing, so it becomes difficult because we clearly don't want to have to ping the channel and bother lots of students at the same time to update their local copies. Ideally, we would like to not have that many fixes after release as well.

But you are right. We will try to go about this better and see what works best to inform everyone and minimizing problems as well.

adammalpass, 2017-10-02:
Thanks for your comments. How about adding an 'Updates' section or something on the README to give a summary of changes. Then everything is all in one place and it would be easy to tell if/when/why we should update our repos.

closed 2017-10-13 by sahiljuneja