udacity/robotics-nanodegree-issues #161: Problems running script follower.py in the Follow Me project

Things2Connect, 2017-10-09:
I have problems running the follower.py script in the Follow Me project (section 6-P-9).  I have renamed the model_weights file (output of simulation) to model_weights.h5 (just added the .h5 extension) and run following line in the RoboND enviroment, after starting the QuadSim simulator and clicking "Follow Me" button + selecting "Spawn People" :  

$ python follower.py model_weights.h5

The python file runs for about 5 seconds then crashes with the Windows message "Python has stopped working".  I am using the following in versions in local mode:

Windows (10) operating system

QuadSim simulator :  v1.2.2

Python : 3.5.2 

Also, I've installed as instructed: 
pip install tensorflow==1.2.1
pip install socketIO-client
pip install transforms3d
pip install PyQt5
pip install pyqtgraph

I am running version

danzelmo, 2017-10-09:
Hmm, that error message sounds scary:) The project has run on multiple windows 10 machines, so that would make me look for a system/build specific problem. 
Here is what I would do.

first: run a helloworld.py type of script from the command line, and make sure this runs. 

second: check if you have multiple different python installs on your system, and that you have the correct version active while running `follower.py`

third:  if you are not getting a stack trace of any kind when running the script. Then I would put an `if False:` about half way down the `if __name__ == '__main__'` section of follower.py, and then indent the rest of the section so it doesn't run. 

I would then adjust the position of this until the line that is causing the problem is found.

sahiljuneja, 2018-04-26:
Closing because of no updates. If this was not resolved then we can reopen this.

closed 2018-04-26 by sahiljuneja