udacity/robotics-nanodegree-issues #164: deep learning 1.24 formula error

General-Sax, 2017-10-18:
Unless I am seriously misunderstanding the notation in this section, the 'summary' formula just above the quiz erroneously includes a subscript i (dE/dw_i); since this derivative equals the contributions from all weights, this should be dE/dW, or a summation over i.

General-Sax, 2017-10-18:
Upon revisiting this, it's not an error, it's just confusing how this is indexed, or where the subscript is placed. This equation is saying that the error gradient contribution from the ith point x,  x^(i), is given by this scalar multiplication on x^(i), but that is not what it looks like. This isn't a true error, just a point that may deserve further clarification/specificity.

danainschool, 2018-05-03:
part of DL content and not a bug - will let them update as needed.

closed 2018-05-03 by danainschool