udacity/robotics-nanodegree-issues #168: Kuka Pick and Place Arm not picking up the cylinder, even in demo mode

PatrickMockridge, 2017-10-28:


PatrickMockridge, 2017-10-28:
[image]
[image]
[image]

kylesf, 2017-11-02:
Can you please share the specs of your set up and what version of the VM you are using?

PatrickMockridge, 2017-11-03:
Hi Kyle I am going to reinstall the VM with the very latest version and rebuild my pick and place project in the new VM. When I’ve run the demo I will update you as to how it went.

PatrickMockridge, 2017-11-03:
Hi Kyle,

I reinstalled the VM and set up Project 2 again from scratch. I’m sorry to report that I am still getting the same issues. The gripper is still not gripping the cylinder even in demo mode.

I’ve been around the block with this now so many times. What else can I do? I am sort of reaching my limit of patience.

Best,

Patrick

From: Patrick Mockridge <patrick@engzig.com>
Date: Friday, 3 November 2017 at 19:34
To: udacity/robotics-nanodegree-issues <reply@reply.github.com>
Subject: Re: [udacity/robotics-nanodegree-issues] Kuka Pick and Place Arm not picking up the cylinder, even in demo mode (#168)

Hi Kyle I am going to reinstall the VM with the very latest version and rebuild my pick and place project in the new VM. When I’ve run the demo I will update you as to how it went.

PatrickMockridge, 2017-11-03:
My laptop is a Lenovo Ideapad 510s running Ubuntu 16.04, 8GB RAM, core i5-7200U. My VM is set up in VMware Player version 14.0.0 with 3 cores and 6GB RAM assigned

PatrickMockridge, 2017-11-03:
I just downloaded and reinstalled the latest version of the VM (2.1.0) tonight and reinstalled everything from scratch. Still the same issues.

ancabilloni, 2017-11-06:
Hi @PatrickMockridge which option are you using for the robot routine? "Continue" or "Next". If you're using "Continue", can you try with "Next" for manually tell the robot move on to the next step after it's completely done with the action before it moves on to the next action? Sometimes an action moves quicker than it should and slip out in grabbing the object. Though this can be varied per many cases.

closed 2017-11-21 by kylesf