udacity/robotics-nanodegree-issues #169: Follow Me Simulator Python script crash Win10

Thom087, 2017-11-02:
I launch the simulator in the follow me mode. As soon as I try to launch the python script It takes like 10s than it crashes. I am using Win10

Thom087, 2017-11-02:
[image]

Thom087, 2017-11-02:
The simulator somehow is still working no problem, just the python script won't work

kylesf, 2017-11-02:
Please run `conda update --all` and see if that fixes the problem!

kylesf, 2017-11-06:
Any updates @Thom087 ?

adammalpass, 2017-11-12:
@kylesf @Thom087
I just spoke to student on Live Chat who had this exact same issue. I suggested to run `conda update --all` and it fixed their problem! :)

perfalcon, 2017-11-12:
@kylesf  .. i am facing the same issue, it worked after i did "conda update --all"

adammalpass, 2017-11-17:
Another student on Live Chat had this same issue and again the same fix solved it. So I think this one should be solved!

danzelmo, 2017-11-17:
The update fixes the issue, but I am not sure why yet. Its on my todo list to track down the exact cause of this, I will leave this open until that fix is found.

closed 2018-04-26 by sahiljuneja